Unitree-Go2-Robot/go2_robot #4: Support For Go2 Pro

David9696, 2024-07-05:
Hi,

Thanks for sharing this great project.

Is this package support go2 pro? Or it only supports go2 edu?

If this package does support go2 pro, how could I use it with the go2 pro robot? Via wifi or Ethernet?

Hope you could enlighten me.

Thanks

Regards,

David

Juancams, 2024-07-05:
Hi @David9696 !

I have been reviewing, and I think the only difference they have in go2 EDU vs Pro is that the EDU can come with a built-in jetson orin nx, but the Pro does not... could you confirm this for me?

If that is the only difference, you can launch the driver from your computer, either via Wifi or Ethernet.

I hope I've helped!!

Greetings.

David9696, 2024-07-06:
Hi @Juancams 

Thanks for your reply. I have tested this package with go2 pro robot. It is not working via wifi and ethernet. Most likely it is working only with edu version as nly edu version can be accessed via dds. 

Anyway, thanks for sharing this great project.

Juancams, 2024-07-08:
Hello again @David9696 

I've been checking, and you probably need to do some previous steps to be able to launch it on the Pro. The reason is that there is an sdk that the robot launches by default in the EDU versions, since these come equipped with a jetson.

Before using our packages, please follow the steps in the following repository and get this [sdk](https://github.com/unitreerobotics/unitree_sdk2) running on the computer with which you communicate with the robot. Once you achieve it, launch our driver and you should be able to use it as I indicate in the examples.

Please, if you manage to get it working, tell us so we can update our repository, or make a pull request with the steps to follow, and thus support the rest of the go2 versions.

David9696, 2024-07-12:
Hi @Juancams ,

Thanks for your suggestion. I have tried to install the sdk but still not working. The jetson on edu robot might have some special keys to access the dds from robot.

These are the package and steps that I have taken in case anyone is interested:

1. Install this sdk(https://github.com/unitreerobotics/unitree_sdk2), but still cant access to the go2 pro robot.
2. Follow the setting up steps here(https://support.unitree.com/home/en/developer/Quick_start), but still cant access to the go2 pro robot.

If I missed anything, please let me know.

Anyway, thanks for suggestion.

cyhunblr, 2024-10-24:
> Hi @Juancams ,
>
> Thanks for your suggestion. I have tried to install the sdk but still not working. The jetson on edu robot might have some special keys to access the dds from robot.
>
> These are the [...]

Did you find any kind of solution?

Juancams, 2024-10-24:
Hi @cyhunblr 

I haven't been able to do anything with this, since I have the EDU version and I don't know the keys.

Sorry.

Juancams, 2025-03-24:
I'm closing this issue because there are no more updates, and no solution has been found to this problem. If anyone has the Go2 Pro robot and can help with this, feel free to reopen the issue.

closed 2025-03-24 by Juancams