Unitree-Go2-Robot/go2_robot #6: Caught exception in launch (see debug for traceback): executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffff8a587f40>]' not found on the PATH root@ubuntu:/sdk_ws# ros2 launch go2_bringup go2.launch.py

artelrobotics, 2024-07-05:
[image]

I installed everything with README instruction and having this error when launch it in docker container .
Can you help?

Juancams, 2024-07-05:
Hi @artelrobotics 

I remember that error, but I would have to look at what command was used within the robot. I don't have access to it until Monday, so I won't be able to help you until then :(

In any case, the error indicates that you need to install something that the launcher is using, and I seem to remember that it has something to do with xacro that is using the go2_description. Use the following command:

```bash
sudo apt install ros-$ROS_DISTRO-xacro
```

If this doesn't solve it, also try:

```bash
sudo apt install ros-$ROS_DISTRO-launch-ros
```

Please tell me if any of these commands have helped you or if you have been able to solve it with another one.

Greetings

rikhsitlladeveloper, 2024-07-11:
Yes it helps. Thanks @Juancams
Is any progress with SLAM and Nav2?

Juancams, 2024-07-11:
Hello @rikhsitlladeveloper 

I'm glad it worked.

Regarding slam and nav2, we are in the testing phase of the parameters that we have prepared, and I hope that everything will be ready at the end of the month.

For additional lidars that can be attached to the robot you will have to wait a little longer.

caicaixia, 2024-07-30:
[image]
 I don't have the Hesai Lidar. So I comment some lines in the launch file. However, it can not work!