Unitree-Go2-Robot/go2_robot #10: Feature: Go2 front camera support

mspringer1, 2024-07-12:
Hey,

based on the Go2 SDK Multimedia Service Interface (https://support.unitree.com/home/en/developer/Multimedia_Services) it would be possible to add video from the front camera to the go2_driver as sensor_msgs/CompressedImage.

A reference implementation is given under **Video streaming**
```
#include <opencv2/opencv.hpp>
using namespace cv;

#include <iostream>
using namespace std;

int main()
{

  VideoCapture cap("udpsrc address=230.1.1.1 port=1720 multicast-iface=<interface_name> ! application/x-rtp, media=video, encoding-name=H264 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! video/x-raw,width=1280,height=720,format=BGR ! appsink drop=1", 
            CAP_GSTREAMER);
    
  if (!cap.isOpened()) {
        cerr <<"VideoCapture not opened"<<endl;
        exit(-1);
    }
    
    while (true) {

        Mat frame;

        cap.read(frame);

        imshow("receiver", frame);

        if (waitKey(1) == 27) {
            break;
        }
    }

    return 0;
```

Juancams, 2024-07-17:
Hello @mspringer1 

Following the example shown by unitree, in a few weeks the video from the camera that is integrated into the go2_driver will be implemented.

Regarding how to publish it, we had planned to use image_transport.

mspringer1, 2024-07-18:
Hey @Juancams 

there is also already a ros2 node written in rust for that: https://github.com/tfoldi/ros2_go2_video

Juancams, 2024-07-18:
Yes, I know that code, but the idea is to reimplement it in c++ and not use anything from webrtc.

tfoldi, 2024-11-05:
you can use any gstreamer based ros2 node to add video feed without doing webrtc

tfoldi, 2024-12-19:
added native video support via #24

Juancams, 2025-03-24:
This feature has already been added to this [PR](https://github.com/Unitree-Go2-Robot/go2_driver/pull/3). I'm closing the issue.

closed 2025-03-24 by Juancams

AlexWUrobot, 2025-08-27:
Dear Developers,

May I ask how to subscribe to the image_raw topic from the Unitree Go2?

I have installed and sourced the following package:
https://github.com/unitreerobotics/unitree_ros2

Currently, I can see the following:

```
ros2 topic info /frontvideostream
Type: unitree_go/msg/Go2FrontVideoData
Publisher count: 25
Subscription count: 1

```

And the message structure:

```
ros2 interface show unitree_go/msg/Go2FrontVideoData
uint64 time_frame
uint8[] video720p
uint8[] video360p
uint8[] video180p
```

I also checked the topic frequency:

```
ros2 topic hz /frontvideostream
average rate: 33.777
    min: 0.000s max: 0.113s std dev: 0.03392s window: 34
average rate: 31.029
    min: 0.000s max: 0.113s std dev: 0.03367s window: 64
```

However, when I run my Python script, I am not receiving any messages:

```
python3 go2_camera_viewer.py
running the code......
```

Could you please advise on the correct way to subscribe to the image_raw feed from the Go2 camera?

Thank you!

Here is my code 
`import rclpy
from rclpy.node import Node
from unitree_go.msg import Go2FrontVideoData
import cv2
import numpy as np

class FrontVideoSubscriber(Node):
    def __init__(self):
        super().__init__('front_video_subscriber')
        self.subscription = self.create_subscription(
            Go2FrontVideoData,
            '/frontvideostream',
            self.listener_callback,
            10
        )

    def listener_callback(self, msg: Go2FrontVideoData):
        print("receive msg ")
        # Pick the resolution you want
        if len(msg.video720p) > 0:
            np_arr = np.frombuffer(msg.video720p, dtype=np.uint8)
            frame = cv2.imdecode(np_arr, cv2.IMREAD_COLOR)
            if frame is not None:
                cv2.imshow("Go2 Front Camera (720p)", frame)
                cv2.waitKey(1)
            else:
                self.get_logger().warn("Failed to decode 720p frame")
        else:
            self.get_logger().warn("Empty video frame received")

def main(args=None):
    print("running the code ...... ")
    rclpy.init(args=args)
    node = FrontVideoSubscriber()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()
`

gonzalosanz52-lab, 2026-01-12:
Hi all,

Has anyone been able to get the front camera working with this go2_robot implementation? I see the original feature request for front camera support (#10) but it’s closed and there’s no follow-up on implementation status.

Specifically:

Are there any working ROS2 topics or nodes that actually stream the front camera data?

What topic names, message types or drivers should we be using?

Has someone tested this with a real Go2 and can confirm you’re receiving frames?

If you have a minimal example (launch file / subscriber / config) that works, please share it — this is exactly what’s missing in the current thread.

Thanks!