Unitree-Go2-Robot/go2_robot #11: No internet connection on GO2 Edu

Cavalletta98, 2024-07-15:
Hi, i have a GO2 Edu with a jetson mounted on it. I noticed that is has no internet connection. How do you manange to connect to the robot or installing ROS Humble on it?

Juancams, 2024-07-17:
Hello @Cavalletta98 

To connect it to the internet, I connect the robot to a router through an RJ45.

We have the robot with Ubuntu 20, which is how it comes by default, so we have ros foxy and noetic. In the future we will update the robot's jetpack to be able to use humble or jazzy.

Cavalletta98, 2024-07-17:
Is there a router mounted on the robot?

Juancams, 2024-07-18:
No, the robot by default does not come with a router. You would have to connect it to one you have lying around and then give the network to that router, or you can also put a USB adapter on it to connect it to a network.

mspringer1, 2024-07-18:
In some documentation I read that this model: "TP-Link TL-WN722N High Gain WLAN USB Adapter" works out of the box with the Go2 and is used by others, so no additional drivers need to be installed.

Juancams, 2024-07-18:
The robot can set up its own wifi, but it appears to be connected to another part of the robot, not to the jetson. When I connect and try to see the topics it doesn't work, I thought that modifying the dds could do the same, but in reality the computer does not have any new network interface through which to send the information.

The only way I can think of for now is to connect a tp link or a router to the robot.

Cavalletta98, 2024-07-23:
ok thank you so much

closed 2024-07-23 by Cavalletta98