Unitree-Go2-Robot/go2_robot #14: Robot is falling down

rikhsitlladeveloper, 2024-08-09:
After bringup launch. I am controlling robot using teleop twist keyboard.
It was walking good before. But now it is falling down a lot. It is not stable. It is doing something strange.
But when we control it using joystick everything is good and smooth.
Can you clarify situation? 
Can I change power mode of Jetson to FULL Mode

Juancams, 2024-10-23:
Hello!

I don't understand why it worked fine at first and now it crashes. You'll have to send me a demo (video of the robot and the terminal, if possible) of what exactly happens, because at first I don't know what's happening.

Regards.

Juancams, 2025-01-10:
I'm going to mark this issue as resolved. If it isn't, or a related issue reoccurs, please feel free to reopen it.

closed 2025-01-10 by Juancams