Unitree-Go2-Robot/go2_robot #15: Feature: cmd_vel with active obstacle avoidance

mspringer1, 2024-09-17:
Hey,

I noticed when using the default cmd_vel interface, the native obstacle avoidance of the go2 is not active.

The obstacle avoidance client also offers a move function (https://support.unitree.com/home/en/developer/ObstaclesAvoidClient) so I would propose a second cmd_vel subscriber in the go2_driver with a different topic e.g. `/obstacle_avoid/cmd_vel` with a callback to the move function of `api/obstacle_avoid/request` instead of `api/sport/request`.

This could then be used for manual control of the robot to add more safety.

Juancams, 2024-10-24:
Hi!

`/cmd_vel` is only used to command speeds to the robot, and you already use any algorithm to be able to avoid obstacles, so it is normal that it collides if you go against a wall HAHAHA but what you propose, which is to create another topic that makes the robot not advance against an obstacle even if you command it speed, seems like a good idea to me.

Juancams, 2025-03-24:
I'm closing this issue because there haven't been any further updates. If anyone decides to revive this feature, feel free to reopen it.

closed 2025-03-24 by Juancams