Unitree-Go2-Robot/go2_robot #18: Strange `/cmd_vel` behaviour

tgodfrey0, 2024-10-23:
Hi,

I am trying to control the robot using the `/cmd_vel` topic. But regardless of the message contents, the robot moves forward in a very strange motion (video attached). This even happens if the message is all zeros. Do you have any ideas about this? 

https://github.com/user-attachments/assets/1c7bff28-119a-4a8d-8f6f-f635af398020

Thanks,
Toby

mspringer1, 2024-10-23:
Hey Tobi, maybe there is an issue with your cmd_vel publisher or wireless connection. It seems like the twist message is not received continuously but periodically. Does this also happen when you publish on cmd_vel via ros2 topic on the robot?

Juancams, 2024-10-23:
Hi

I think the problem is with the hz you are posting at. Try posting at a rate of 20 or 30 hz, using `--rate 30`. If this is not the problem, please let me know and we can see what else might be going on.

Regards.

tgodfrey0, 2024-10-24:
Hi,

If I publish from the robot at either 20 or 30 Hz, the motion is a lot smoother. But the robot still moves forward regardless of message content (positive linear x, negative linear x, non-zero angular z, all zeros).

Thanks

mspringer1, 2024-10-24:
Could you verify that you use the correct formatting when sending a twist message via the ros2 topic cli? Especially for the message content? Do you use completions when you enter the message content? Sometimes a wrong format can lead to wrong values for the velocities.

Could you try sending cmd_vel via rqt_robot_steering?

tgodfrey0, 2024-10-24:
This is what I am calling

```bash
ros2 topic pub --rate 50 /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}"
```

I get the same behaviour with `rqt_robot_steering` and `teleop_twist_keyboard`

tgodfrey0, 2024-10-24:
I have fixed this now. It was interference from another node on the network. Thanks for the help with the rate anyway

closed 2024-10-24 by tgodfrey0