Unitree-Go2-Robot/go2_robot #19: Pre-installed ROS2 version

figkim, 2024-10-28:
Hello,

Thank you for sharing such useful code. I am currently operating a Go2 EDU system, and the Jetson NX in the expansion dock is set up with Foxy and Noetic. The README assumes Humble is being used. Did you write the README assuming that Jetson NX is being used as well, or was it written with a different device in mind?

If it’s the former, did you install Humble on the Jetson NX in addition?

Thank you for your clarification.

Juancams, 2024-10-29:
Hi!

The README is designed to run everything on your computer, so I assume you are using ros 2 humble. Soon it will also be available for the jazzy and possibly rolling version.

Regarding the robot, everything is by default, so it is using ros 2 foxy. You can run the code you find here inside your robot using foxy and it would work, but this package is not intended to be used inside, especially on higher versions.

Regards.

figkim, 2024-10-29:
Thank you for your reply. I also proceeded under the assumption of installing ROS Humble on a separate computer (host). After completing the installation and running `ros2 launch go2_bringup go2.launch.py`, I encountered an error stating that the Hesai Lidar ROS package could not be found. When I followed the sensor installation steps for the Hesai Lidar (as per the README) in the Humble environment on the host PC, I ran into build errors.

In the README, there’s a section that says, “If you have purchased a Hesai Lidar 3D or a Realsense D435i, follow the following steps inside the robot.” 
Does this mean that the sensor installation should be performed inside the robot from this point onwards? not in host?

Thank you

Juancams, 2024-10-29:
Yes, this is a bit confusing, to be honest, since we have also adapted the launcher so that it can be launched inside the robot and you can launch the extra sensors that you equip, although as I mentioned it is mainly prepared so that you can launch it from the host.

I have to review and accept #8  to make it so that it does not ask you to have hesai or realsense installed.

figkim, 2024-11-19:
Hello. I finally succeeded in setting it up. The environment is an Ubuntu 20.04 host running an osrf:humble container, and there’s no separate ROS installation on the host. The host is connected via LAN cable to the Go2 Edu’s Orin NX, and I successfully performed the bringup. I confirmed that the basic LiDAR data is being received properly, and when I move the robot with a joystick, I can see the changes in tf (transform frames) in RViz.

However, while the data sent from the Go2 to the host is confirmed and working well, it seems that messages sent from the host to the Go2 are not functioning properly. For example, I tried the `service for “hello” mode` and attempted to move the Go2 by publishing to the `/cmd_vel` topic, but both attempts failed. Could you advise me on where else I should look?

Juancams, 2024-11-19:
Hello!

What you are telling me is a bit strange. I cannot confirm where I have come from to test the `"hello" service` and others, but I have been using navigation2, which in the end publishes in `/cmd_vel`, launched from my computer, and it has worked correctly.

Are you launching the robot bringup inside or outside the robot?

figkim, 2024-11-20:
Hello. As described above, I performed the bringup from outside the robot (within the `osrf:humble` container on an Ubuntu 20.04 host). While using this repository, I have never directly connected to the robot via SSH. When I executed the bringup, the output was as shown below, and I confirmed that RViz is working well as displayed in the README.

```
root@user-Nuvo-8108GC-Series:~/go2_ws# ros2 launch go2_bringup go2.launch.py rviz:=True
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-11-20-11-32-12-620292-user-Nuvo-8108GC-Series-908
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [921]
[INFO] [component_container-2]: process started with pid [923]
[INFO] [rviz2-3]: process started with pid [925]
[rviz2-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[robot_state_publisher-1] [WARN] [1732069932.856381859] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-1] [INFO] [1732069932.856498905] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1732069932.856539857] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1732069932.856546932] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1732069932.856551921] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1732069932.856556389] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1732069932.856560658] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1732069932.856564958] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1732069932.856569281] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1732069932.856573673] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1732069932.856578003] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1732069932.856582357] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1732069932.856586677] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1732069932.856591191] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1732069932.856595552] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1732069932.856599817] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1732069932.856604113] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1732069932.856608545] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1732069932.856612932] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1732069932.856617273] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1732069932.856621576] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1732069932.856627857] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1732069932.856635370] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1732069932.856642442] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1732069932.856650454] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1732069932.856658579] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1732069932.856666157] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1732069932.856673931] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1732069932.856681489] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1732069932.856686660] [robot_state_publisher]: got segment radar
[component_container-2] [INFO] [1732069933.068644220] [go2_container]: Load Library: /root/go2_ws/install/go2_driver/lib/libgo2_driver.so
[component_container-2] [INFO] [1732069933.073087979] [go2_container]: Found class: rclcpp_components::NodeFactoryTemplate<go2_driver::Go2Driver>
[component_container-2] [INFO] [1732069933.073106204] [go2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<go2_driver::Go2Driver>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/go2_driver' in container '/go2_container'
[rviz2-3] [INFO] [1732069933.220891938] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1732069933.221028958] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1732069933.271294897] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [ERROR] [1732069933.994809996] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-3]
[rviz2-3] [ERROR] [1732069933.995154342] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
```

[image]

Regarding the service, I tried executing the following command after the bringup, as guided in the README, but there was no response from the robot. (I expected it to greet as shown on the main page of this repo, but there was no action.)

`ros2 service call /mode go2_interfaces/srv/Mode "mode: 'hello'"`

Also, after the bringup and once the /cmd_vel topic was created, I performed the following topic publish via CLI. I expected the robot to move forward, but again, there was no noticeable action.

```
Input:
ros2 topic pub --rate 10 /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.5}, angular: {z: 0.0}}"

Then Output:
publishing #376: geometry_msgs.msg.Twist(
    linear=geometry_msgs.msg.Vector3(x=0.5, y=0.0, z=0.0), 
    angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=0.0)
)
```