Unitree-Go2-Robot/go2_robot #21: SLAM & Nav2 Availability

RiccardoZuppetti, 2024-11-11:
Hi, I saw that both Slam and Navigation features are currently being worked on, will they be available in the next few weeks?

Juancams, 2024-11-11:
Hello @RiccardoZuppetti 

Yes, you could say that it is finished, just needing to give it a few more tests. Everything is in #20 , which I encourage you to try and give us your opinion on whether it works correctly.

It says foxy but it can be tested in humble and jazzy versions, as long as it is launched from your computer, or launched inside the robot in its foxy version.

Greetings.

RiccardoZuppetti, 2024-11-12:
Hi @Juancams

Thanks for the prompt feedback, I'll definetly give a try to the content of the PR and I'll give to you my feedback usage as well. By the way, I've noticed that still there is not a config for the slam feature, it will come after these final tests?

Juancams, 2024-11-12:
For slam you would have to use the same launcher configurations that were made for browsing, and regarding the params file, you would only have to indicate the topics and frames.

We will try to have everything uploaded at the same time.

mebha, 2024-12-25:
Hi, could you please tell me how to implement SLAM and nav2? Thanks in advance. 
Also,could you tell me when you the SLAM and Nav2 part will available?

Juancams, 2025-01-10:
Hello!

Nav2 is available, but we haven't uploaded a file for Slam. We will do some tests and add it in the next few days.

Regards.

nathanielsade, 2025-03-04:
hi does the nav2 and slam 
works ?

Juancams, 2025-03-04:
Although there is no slam package, you can use the generic one, adding the topic you are going to use for laser and the frame_id of the robot.

Nav2 works correctly.

judy33a, 2025-03-04:
@nathanielsade connect ur pc and the robot via ethernet and run the code on the robot (clone the repo on the go2), run rviz2 on the pc if you want to visualize your data/robot, (note that the distro on the go2 is ros2 foxy in case u need to install missing libraries)

nathanielsade, 2025-03-18:
hey
+ @judy33a 
were having this issue when trying to run foxy-devel version as you said:

unitree@ubuntu:~/go2_ws$ ros2 launch go2_bringup go2.launch.py
[INFO] [launch]: All log files can be found below /home/unitree/.ros/log/2025-03-18-17-18-23-404763-ubuntu-79315
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffffb1191ee0>]' not found on the PATH")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 132, in evaluate_parameter_dict
    evaluated_value = value.evaluate(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/parameter_descriptions.py", line 83, in evaluate
    self.__evaluated_parameter_value = perform_typed_substitution(
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/type_utils.py", line 548, in perform_typed_substitution
    perform_substitutions(context, cast(List[Substitution], value)),
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/command.py", line 94, in perform
    command_str = perform_substitutions(context, self.command)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_context.py", line 232, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/path_join_substitution.py", line 45, in perform
    performed_substitutions = [sub.perform(context) for sub in self.__substitutions]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/path_join_substitution.py", line 45, in <listcomp>
    performed_substitutions = [sub.perform(context) for sub in self.__substitutions]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/substitutions/find_executable.py", line 66, in perform
    raise SubstitutionFailure("executable '{}' not found on the PATH".format(self.name))
launch.substitutions.substitution_failure.SubstitutionFailure: executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0xffffb1191ee0>]' not found on the PATH

we would really appriacte your support.

nathanielsade, 2025-03-18:
> Although there is no slam package, you can use the generic one, adding the topic you are going to use for laser and the frame_id of the robot.
> 
> Nav2 works correctly.

Hi, thanks for your answer.

I'm new to the robotics world. Could you please be more specific?

Which topics should I add, and where, to enable the robot to navigate? I know that I need to create a map inside RViz, and after that, I can send it a target.

Juancams, 2025-03-18:
I'm going to leave you [here](https://docs.nav2.org/tutorials/docs/navigation2_with_slam.html) with one of the most typical tutorials for slam, I hope it helps. If you find it too difficult, I'm working these weeks on creating a [documented website](https://unitree-go2-robot.github.io/) with everything dog-related, including a slam section.

Juancams, 2025-04-03:
A version of the package, not yet final, for slam has just been uploaded to this [repository](https://github.com/Unitree-Go2-Robot/go2_slam/tree/humble-devel). For now, with a few minor modifications discussed in this [issue](https://github.com/Unitree-Go2-Robot/go2_robot/issues/32#issuecomment-2774811526), and everything in the development branches, slam can be implemented easily.

When everything is merged into the stable branches, we will notify and close this issue.

Regards ^^

belbo9, 2025-07-13:
Hi @Juancams 
I hope you are well mate. Just wanted to ask for an update on your final version of SLAM if possible?
Thanks you.

JeewanthaSadaruwan, 2026-01-22:
thank you very much for your reply. I really appreciate it. Yes, could you please share the migration procedure? That would be extremely helpful.
Just to give you some context, I contacted Unitree directly and they explicitly advised not to install any Ubuntu version other than 20.04 on the robot. Given that recommendation, I am being cautious about upgrading to Ubuntu 22.04.

natxodeuser, 2026-01-22:
Yes, you're right, if we migrate to version 22.04 you'll lose support from Unitree. I'm very happy with it for now. Also, for example, if you have the Jetson 40-tops, you have a max mode that goes up to 67 tops. I've attached my email address (nvarona@deuser.es), and if you send me an email, I'll attach the document I made. It's in Spanish, but it's easy to understand.

natxodeuser, 2026-01-22:
Guys, I'm trying unsuccessfully to get the Go2 to navigate autonomously from point A to point B. I've looked countless times at the parameters in a simple nav2_params.yaml file, which I've shared. But I can't get it to accept goals. When it arrives, the motors jerk, and then it starts circling until it times out and says "Goal fail" or refuses to accept any destination. I'm using the base from this repo to control the robot, and I've created a joystick node that works perfectly, but I can't get it to perform even the minimum distance correctly. I've also noticed that the Go2's MCU is giving an incorrect timestamp compared to the integrated Jetson. Can anyone shed some light on this? Or do you have a better, optimized nav2_params file than mine?

`amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.05
    alpha2: 0.1
    alpha3: 0.05
    alpha4: 0.05
    alpha5: 0.05
    base_frame_id: base_link
    odom_frame_id: odom
    global_frame_id: map
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: False
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 12.0
    laser_min_range: 0.1
    laser_model_type: "likelihood_field"
    max_beams: 180
    max_particles: 8000
    min_particles: 2000
    pf_err: 0.01
    pf_z: 0.95
    recovery_alpha_fast: 0.3
    recovery_alpha_slow: 0.01
    resample_interval: 2
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: True
    transform_tolerance: 5.0
    update_min_a: 0.05
    update_min_d: 0.05
    z_hit: 0.85
    z_max: 0.05
    z_rand: 0.05
    z_short: 0.15
    scan_topic: scan
    set_initial_pose: False
    initial_pose_x: 0.0
    initial_pose_y: 0.0
    initial_pose_z: 0.0
    initial_pose_a: 0.0

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 100
    default_server_timeout: 20
    transform_tolerance: 2.0
    default_nav_to_pose_bt_xml: "/opt/ros/humble/share/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml"
    default_nav_through_poses_bt_xml: "/opt/ros/humble/share/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml"  # Opcional, pero para consistencia
    # Simplificado - solo lo esencial
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_compute_path_through_poses_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_remove_passed_goals_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
[... 221 lines cut ...]
  ros__parameters:
    use_sim_time: False
    loop_rate: 20
    stop_on_failure: False
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: False
    smoothing_frequency: 10.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.5, 0.0, 1.0]
    min_velocity: [-0.5, 0.0, -1.0]
    max_accel: [0.6, 0.0, 1.2]
    max_decel: [-0.6, 0.0, -1.2]
    deadband_velocity: [0.05, 0.0, 0.05]
    odom_topic: "odom"
    odom_duration: 0.1
    velocity_timeout: 1.0
`