Unitree-Go2-Robot/go2_robot #22: Can't access nodes over the network

tgodfrey0, 2024-11-12:
Hi,

When I run the bringup code, I cannot see the nodes on the network. If I SSH into the robot and run it I can control the robot, view it in Rviz2, etc. However, I cannot access the nodes elsewhere on the network. I have tried it over ethernet and over WiFi (with a USB WiFi adapter) but cannot see them. Has anyone else had this issue? It might be a CycloneDDS configuration issue but I am not sure. 

Thanks

Juancams, 2024-11-12:
Hi @tgodfrey0 

You probably have to configure cyclone in an xml file to make the data available through other interfaces, since I think the default configuration of the robot only allows it to see the data through its eth.

tgodfrey0, 2024-11-13:
Hi,

I've looked at the `cyclonedds.xml` file on the robot and it already has `eth0` as a listed interface. Did you have to add another wired interface here?

Juancams, 2024-11-13:
Hi!

We don't have any xml configuration with that right now, but I encourage you to try to make one that works over wifi or both interfaces and share it here.

Regards.

tgodfrey0, 2024-11-27:
I've got this working over ethernet. Even after enabling the wireless interface in the `cyclonedds.xml` file on the Go2, the nodes are still only pushed over `eth0`. I'll keep working on getting wireless working. 

I've also created a Dockerfile that automatically configures everything to allow for easier use of this SDK.

judyakl, 2024-12-05:
> I've got this working over ethernet. Even after enabling the wireless interface in the `cyclonedds.xml` file on the Go2, the nodes are still only pushed over `eth0`. I'll keep working on getting [...]

Hello, did you manage do make it work over wireless network? i also tried adding wireless interface in cyclonedds.xml but it did not work

tgodfrey0, 2024-12-05:
I have this working over wireless now. But I am having an issue with the SDK, even when run on the Jetson, the topics from this package, e.g. `/joint_states` exist but do not publish any messages. Do you have any ideas about this @Juancams? Thanks 

@judy33a I had to add `wlan0` to the `cyclonedds.xml` file, but due to my insitution's network setup I have to run the Go2 wifi adapter in AP mode to create it's own WiFi network. I have  Dockerfile which *should* just work when you build it with the wireless interface.  My fork is [here](https://github.com/tgodfrey0/go2_robot/tree/humble/auxiliary).

closed 2024-12-06 by tgodfrey0

tgodfrey0, 2024-12-07:
I'm not sure of the exact model, it's an old linksys one we had in the lab.

wilhelmsteve92, 2025-05-15:
@tgodfrey0   I read you struggled publishing the internal topics over WiFi. I have the same issue at the moment. Were you able to resolve this problem? I looked over your auxiliary package but I am not quite sure if that or what exactly was the solution for internal topics.

tgodfrey0, 2025-05-16:
@wilhelmsteve92 I had to add a USB WiFi adapter and then make sure the `cyclonedds.xml` file on the Go2 has the WiFi interface selected. After this, it worked for the topics that are provided by this package. The internal topics from other controllers on the robot are not published over WiFi; I'm not sure if there is anything we can do about that because they are sent from the internal controller which Unitree do not provide access to. I guess if you really need the internal topics, you could write a simple replublisher node which runs on the Jetson to push them over WiFi.

Juancams, 2025-05-16:
Hi @tgodfrey0 @wilhelmsteve92 

In my case, we added a nano router to receive the topics published by both this package and the bot. The only configuration I've done is cyclonedds.xml.

wilhelmsteve92, 2025-05-16:
@Juancams so, you mean connecting a nano router to Go2's RJ45 port and then you were able to publish internal topics as well? (cyclonedds.xml config with eth0)

wilhelmsteve92, 2025-05-19:
@Juancams I tried setting up a nano router, connected to Go2's Ethernet port. Tried to different models. I was able to publish internal topics via WiFi but the connection was really bad. High package loss, connection aborted after a few seconds. Did you encounter similar issues? How did you configure your nano router, AP, Client? Which router did you use?

wilhelmsteve92, 2025-05-20:
@tgodfrey0 I looked over go2_robot/auxiliary/ again. At the beginning you say: "The topics from this package can be published over WiFi." and in the section "ROS over WiFi" you say: "Now you should be able to connect to the Go2's network on your PC, open a terminal and run ros2 topic list and see the topics from this package, e.g. /cmd_vel". 
But you mentioned here "The internal topics from other controllers on the robot are not published over WiFi".

So, in my case I am not able to receive cmd_vel and joint_states over WiFi. I can see the topics but nothing is published. All other topics such as LiDAR work. Were you actually able to publish these topics over WiFi using the go2_driver package in go2_robot?

Sorry, for the confusion. As mentioned in my last post, I tried settung up two different nano router connected to Go2's Ethernet port and were able to publish these internal topics. But it wasn't stable; the connection dropped several times after a few seconds, or there was a lot of packet loss. I don't know what I'm doing wrong, or if it's the Go2's network configuration that's preventing a stable connection over the Ethernet port.

tgodfrey0, 2025-05-20:
@wilhelmsteve92 I had `cmd_vel` and `joint_states` working too. I had to put the Go2s WiFi adapter in AP mode and then join that network, but I have heard from other people that it works just by joining another network with the Go2s adapter --- just not at my university

wilhelmsteve92, 2025-05-20:
@tgodfrey0 Just to make sure. 
1. Connect to Go2 via App (AP mode or WiFi).
2. Connect my PC to the same network.
3. Configure cyclonedds on Go2 to publish over both eth0 and wlan0. 
?

No, need to use USB-WiFi-adapter? Or do I need to join the same network as above with my USB Wi-Fi adapter (which is also wlan0)  ?

Thanks for your help?

tgodfrey0, 2025-05-20:
@wilhelmsteve92 What I did:

1. Plug a USB WiFi adapter into the Go2 Jetson and put it into AP mode. Then make a WiFi hotspot from that adapter.
2. Edit `cyclonedds.xml` and add the WiFi interface as a `NetworkInterface`
3. Restart the ROS2 daemon
4. Connect your PC to the Go2's WiFi network
5. Launch the Go2 bringup 

The topics should then appear on the PC (make sure your PC is set to use CycloneDDS over WiFi). I never used the app

wilhelmsteve92, 2025-05-28:
@tgodfrey0 I don't know what I am doing wrong. I am not able to publish internal topics such as joint_states via WiFi. Here is what I do:
Added the WiFi Interface in cyclonedds.xml
Restart ROS2 daemon.
Launch the program.

I can see joint_states etc, but echoing them is not possible.
Even if I run the application just on Go2 (while having eth0 and wlan0 in cyclonedds), the internal topics are not published. If I remove wlan0 in cyclonedds, everything works.

I also resetted the docking station with a fresh factory image, several times. But not working. The weird thing is, I am sure what is was working at least once.

Do I need to configure cyclonedds DOMAIN Id or ROS_DOMAIN_ID or something else?
Unitree support wasn't able to help me yet.