Unitree-Go2-Robot/go2_robot #23: Frame [odom] does not exist

phamthaihoangtung, 2024-12-10:
I'm new to rviz and ros2, after I run `ros2 launch go2_bringup go2.launch.py rviz:=True`, In the `Global Status` > `Fix Frame`, I get the following errors:
`Frame [odom] does not exist`
How can I solve it? Thank you.

judy33a, 2024-12-10:
@phamthaihoangtung switch from odom to base_link, does rviz now show any part of the robot?

phamthaihoangtung, 2024-12-10:
@judy33a it only shows base_link and head_lower and head_upper
[image]

judy33a, 2024-12-10:
@phamthaihoangtung this happened to me as well, it is related to the network connection so no worries since it shows some parts in the base_link,
what i did was that i connected both my laptop and the robot to the same wireless network (while keeping the ethernet) (i already had configured a lot of stuff, so im not quite sure what would be different between my case and yours)

Juancams, 2025-01-10:
I'm going to mark this issue as resolved. If it isn't, or a related issue reoccurs, please feel free to reopen it.

phamthaihoangtung, 2025-01-11:
Finally, I fixed it. After switching from `odom` to `base_link`, I add node `joint_state_publisher` in `go2_description/launch/robot.launch.py`.
```python
    nodes.append(Node(
        package='joint_state_publisher',
        executable='joint_state_publisher'
    ))
```