Unitree-Go2-Robot/go2_robot #26: Lidar-ROS compatibility

Sayantani-Bhattacharya, 2025-01-20:
The lidar package used is not compatible with ROS2 Humble, is there a workaround to using it? I tried replacing it with a different Lidar driver pkg (https://github.com/GITAI/hesai_ros_driver) but it also didn't work for me.
-Thanks.

Juancams, 2025-01-21:
Hello!

I don't think we've checked lidar compatibility with any other version than foxy, since it's the ros 2 that the robot has installed. If you're using a docker inside it, you'll need another version as you say. We're going to leave this issue open so people know and in the meantime we'll try to find a solution, either using the one you've provided us, or trying to make the official one work in other versions.

Thank you very much ^^

Keegsy, 2025-03-11:
@Sayantani-Bhattacharya I've used this repository in the past: https://github.com/HesaiTechnology/HesaiLidar_ROS_2.0
I haven't yet tested it onboard the Go2 but I know it works with ros humble.

Juancams, 2025-03-24:
I am closing this issue assuming that the proposed solution works.

closed 2025-03-24 by Juancams