Unitree-Go2-Robot/go2_robot #30: Bug: Robot Preview in Rviz not proper

driven-by-code, 2025-02-27:
### Check this before filling out

- [x] I have verified that this error has not been reported by another user.
- [x] I have tried to perform the installation again, following the steps.

### Problem location

My Computer

### ROS 2 Version

Humble 🐢

### Package error

- [x] go2_bringup
- [ ] go2_description
- [ ] go2_driver
- [ ] go2_hardware
- [ ] go2_interfaces
- [ ] go2_navigation
- [ ] go2_robot
- [ ] go2_rviz
- [ ] go2_simulation
- [ ] go2_tts
- [ ] go2_tts_interfaces
- [ ] go2_voxel_map
- [ ] unitree_api
- [ ] unitree_go

### What happened?

Hello Everyone,
A bug happened and I am not able to see the robot's preview in Rviz. 
I am connected to the robot with the lidar driver launched on the robot itself.
I have attached an screenshot of rviz. Please let me know if any more information is required. 
Thank you . 

[image]

### Relevant log output

```shell

```

### Reproduction steps

```bash
1.
2.
3.
...
```

Juancams, 2025-02-27:
Hi!
Could you please show me your log so I can see what's going on?

Also, make sure you can see the topics from your computer.

driven-by-code, 2025-02-27:
Hello , What topics would have to be seen on the host pc from the robot ? I can only see the lidar topics on the PC.
I can see all the topics on the Orin NX that is attached to the Robot. 
The log from running the package is enough or do you need anything else ? 
Thanks !

Juancams, 2025-02-27:
Well, you should see the same topics that you see inside the nx... but it's strange that you can see the lidar and not the rest inside your computer... maybe the error has to do with something in the network configuration of your computer with the robot, or maybe with the dds

Are you using cyclonedds on your computer?

driven-by-code, 2025-03-12:
Hello , 
My apologies for the late response.
I sourced cycloneDDS and I am able to see the robot topics.
Here is the log  , pasted below , as requested,
`[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-03-12-12-32-21-862414-dell-pc-923981
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [923995]
[INFO] [component_container-2]: process started with pid [923997]
[INFO] [pointcloud_to_laserscan_node-3]: process started with pid [923999]
[INFO] [rviz2-4]: process started with pid [924001]
[robot_state_publisher-1] [WARN] [1741797142.026303075] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-1] [INFO] [1741797142.026387692] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1741797142.026411525] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1741797142.026417717] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1741797142.026425110] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1741797142.026430808] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1741797142.026435366] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1741797142.026439877] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1741797142.026444194] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1741797142.026448848] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1741797142.026453452] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1741797142.026457746] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1741797142.026462158] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1741797142.026466537] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1741797142.026470921] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1741797142.026475097] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1741797142.026479364] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1741797142.026483887] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1741797142.026488317] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1741797142.026493057] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1741797142.026498961] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1741797142.026503780] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1741797142.026508464] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1741797142.026513367] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1741797142.026517992] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1741797142.026522655] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1741797142.026527180] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1741797142.026531674] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1741797142.026536332] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1741797142.026541104] [robot_state_publisher]: got segment radar
[rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[component_container-2] [INFO] [1741797142.251407482] [go2_container]: Load Library: /home/user/go2_ws/install/go2_driver/lib/libgo2_driver.so
[component_container-2] [INFO] [1741797142.257584538] [go2_container]: Found class: rclcpp_components::NodeFactoryTemplate<go2_driver::Go2Driver>
[component_container-2] [INFO] [1741797142.257620817] [go2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<go2_driver::Go2Driver>
[component_container-2] /opt/ros/humble/lib/rclcpp_components/component_container: symbol lookup error: /home/user/go2_ws/install/go2_driver/lib/libgo2_driver.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_service_type_support_handleIN14go2_interfaces3srv10BodyHeightEEEPK29rosidl_service_type_support_tv
[ERROR] [component_container-2]: process has died [pid 923997, exit code 127, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=go2_container -r __ns:=/'].
[rviz2-4] [INFO] [1741797142.340504799] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1741797142.340591464] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1741797142.354577459] [rviz2]: Stereo is NOT SUPPORTED`

Thank you !

Juancams, 2025-03-28:
I don't know the cause of your error right now, nor have I been able to replicate it. Check that you have all the packages updated, that all the dependencies are installed... I'll keep trying to find the cause of your problem.

glianghan, 2025-05-07:
I encountered the same problem. Have you solved it?

Keegsy, 2025-05-18:
Likewise I cannot get the model to appear correctly - I get the same image as @driven-by-code presented. I am running this in the foxy-dev version. Could that be the issue? @driven-by-code what distro are you using?

Keegsy, 2025-05-18:
Edit: I found mine now works because I accidentally commented out the go2_driver declaration in the launch file.

Juancams, 2025-05-22:
Oh wow, I'm glad you finally found your mistake.

I'm marking the issue as complete.

Regards ^^

closed 2025-05-22 by Juancams