Unitree-Go2-Robot/go2_robot #33: Using Different LiDAR Sensors

umsh315, 2025-03-24:
### Check this before filling out

- [x] I have checked that this question has not been previously resolved.

### What is your question? What do you need help with?

Hello. I am new to ROS2, and I apologize if this is a basic question. Thank you for the wonderful package. I have a few questions:

Is this package already complete with regards to nav2 and SLAM? Also, are there any demo videos available?

I have a Livox ID-360 and would like to use it. Would it be applicable if I change the driver, topic names for navigation, and LiDAR coordinate transformations? If there are any concerns, please let me know.

Thank you in advance.

Juancams, 2025-03-24:
Hi!

Thank you very much for the support. Don't worry if your question is too basic or not; we've all had to start at some point and we have to learn little by little.

In this organization, you can find the package to launch [navigation](https://github.com/Unitree-Go2-Robot/go2_navigation/tree/foxy-devel), but not slam. To launch navigation, you'll have to make the modifications you indicate, that is, perhaps specify the frame, lidar topic, etc. The same would apply to slam; using the [slam_toolbox](https://github.com/SteveMacenski/slam_toolbox) package, you'll just need to load the necessary parameters for it to work. 

We're working on more extensive [documentation](https://unitree-go2-robot.github.io/), where you'll find how to launch Nav2 and Slam, and we'll also create a Slam package with all the parameters already configured, but it will take a couple of weeks. We're also planning to upload videos of how it works.

I hope all of this helps you, and don't hesitate to give us feedback with videos of how it works or ask us any other questions related to this or any other topic.

Best regards

umsh315, 2025-03-26:
Thank you for your response. I plan to take my time and learn from the basics. Having detailed documentation in addition to the code would be very helpful, and I hope it will be completed.

closed 2025-03-26 by umsh315