Unitree-Go2-Robot/go2_robot #34: How to solve the tf transformation in your program?

HangyuDemo, 2025-03-28:
### Check this before filling out

- [ ] I have checked that this question has not been previously resolved.

### What is your question? What do you need help with?

Currently our team want to add slam/navigation to the Unitree go2. While the Go2 official document do provide the URDF to do TF. But they still miss the odom->base_link. We only have hesai lidar without robosense camera. So, I will be very appreciate if you could share some working experience about this part.

Juancams, 2025-03-28:
Hello!!

When you launch the [go2_driver](https://github.com/Unitree-Go2-Robot/go2_driver) package, you'll see the odom->base_link transformation. Also, we have [go2_navigation](https://github.com/Unitree-Go2-Robot/go2_navigation/tree/humble-devel) in this repository, so there's no need to create a new one, just debug it.

Greetings

HangyuDemo, 2025-03-31:
Hi!

We saw the odom -> base_link transformation you implemented under the go2_driver package — thanks for that!

I just wanted to ask if you also provide a Python version of this transformation?

By the way, this is a really great project — full support from our side!

Best!

Juancams, 2025-03-31:
No, in principle everything is C++, but you are free to migrate it and we can open a repository for you to upload it.

I assume your problem is solved, so I'll close the issue.

Regards ^^

closed 2025-03-31 by Juancams