Unitree-Go2-Robot/go2_robot #35: rviz bringup error.

Raviteja-T, 2025-03-31:
### Check this before filling out

- [ ] I have checked that this question has not been previously resolved.

### What is your question? What do you need help with?

Hi, my system environment Ubuntu 20 and ros2 foxy. I commented the lidar and realsense lines in bringup code. when run the ros2 launch go2_bringup go2.launch.py rviz:=True. robot is not visible clearly. 

[image]

Juancams, 2025-03-31:
Hi!

The error probably has to do with the connection to the robot, or you're not receiving the topic information because the DDS isn't configured correctly.

There's a similar [issue](https://github.com/Unitree-Go2-Robot/go2_robot/issues/32) that addresses this topic, so I recommend reading it and seeing if that helps. If not, please follow the thread there.

Regards ^^

closed 2025-03-31 by Juancams

Keegsy, 2025-05-13:
I'm also having the exact same issue with same setup. I'm running the launch file from inside the ssh'd terminal of the go2 robot, but the model doesn't display in rviz:

ros2 launch go2_bringup go2.launch.py rviz:=True
[INFO] [launch]: All log files can be found below /home/unitree/.ros/log/2025-05-13-17-52-52-908842-ubuntu-31447
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [31450]
[INFO] [rviz2-2]: process started with pid [31452]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-1] Link FL_hip had 1 children
[robot_state_publisher-1] Link FL_thigh had 1 children
[robot_state_publisher-1] Link FL_calf had 2 children
[robot_state_publisher-1] Link FL_calflower had 1 children
[robot_state_publisher-1] Link FL_calflower1 had 0 children
[robot_state_publisher-1] Link FL_foot had 0 children
[robot_state_publisher-1] Link FR_hip had 1 children
[robot_state_publisher-1] Link FR_thigh had 1 children
[robot_state_publisher-1] Link FR_calf had 2 children
[robot_state_publisher-1] Link FR_calflower had 1 children
[robot_state_publisher-1] Link FR_calflower1 had 0 children
[robot_state_publisher-1] Link FR_foot had 0 children
[robot_state_publisher-1] Link Head_upper had 1 children
[robot_state_publisher-1] Link Head_lower had 0 children
[robot_state_publisher-1] Link RL_hip had 1 children
[robot_state_publisher-1] Link RL_thigh had 1 children
[robot_state_publisher-1] Link RL_calf had 2 children
[robot_state_publisher-1] Link RL_calflower had 1 children
[robot_state_publisher-1] Link RL_calflower1 had 0 children
[robot_state_publisher-1] Link RL_foot had 0 children
[robot_state_publisher-1] Link RR_hip had 1 children
[robot_state_publisher-1] Link RR_thigh had 1 children
[robot_state_publisher-1] Link RR_calf had 2 children
[robot_state_publisher-1] Link RR_calflower had 1 children
[robot_state_publisher-1] Link RR_calflower1 had 0 children
[robot_state_publisher-1] Link RR_foot had 0 children
[robot_state_publisher-1] Link imu had 0 children
[robot_state_publisher-1] Link radar had 0 children
[robot_state_publisher-1] [INFO] [1747129973.161715402] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1747129973.161905231] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1747129973.161931312] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1747129973.161948720] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1747129973.161962800] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1747129973.161976625] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1747129973.161990257] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1747129973.162004018] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1747129973.162017714] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1747129973.162031154] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1747129973.162045139] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1747129973.162058579] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1747129973.162071507] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1747129973.162084980] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1747129973.162097684] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1747129973.162110484] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1747129973.162123157] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1747129973.162135541] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1747129973.162148822] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1747129973.162161462] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1747129973.162174294] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1747129973.162187223] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1747129973.162200727] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1747129973.162213463] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1747129973.162228952] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1747129973.162242968] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1747129973.162255897] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1747129973.162268441] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1747129973.162282201] [robot_state_publisher]: got segment radar
[rviz2-2] [INFO] [1747129973.519410751] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1747129973.519861963] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[... cut ...]
[rviz2-2] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
[rviz2-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-2] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

[image]