Unitree-Go2-Robot/go2_robot #36: Cannot control robot

16stelter, 2025-05-16:
### Check this before filling out

- [x] I have checked that this question has not been previously resolved.

### What is your question? What do you need help with?

Hello,

first of all, thank you for developing this package, it looks very promising so far. I have followed the readme and got to the point where the rviz visualization starts and I can see the robot tf and point cloud, but I am unable to control the robot. I have tried sending a nav goal through rviz or directly sending a message to /cmd_vel. In both cases the robot does not move. It also seems like the robot does not accept any commands from the remote control after the launch script is started.

I also get a series of warnings from cyclonedds due to the version mismatch between the robot (foxy) and my device (humble):
`[WARN] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/utlidar/mapping_cmd' with type 'std_msgs::msg::dds_::String_' from USER_DATA '(null)'. (handle_builtintopic_endpoint() at /home/sebastian/ros2ws/src/unitree_go2_nav/unitree_ros2/unitree_ros2/cyclonedds_ws/src/rmw_cyclonedds/rmw_cyclonedds_cpp/src/rmw_node.cpp:937`

But since you said you've been running the code with foxy on the robot and humble on the external device, that should be fine, right? Especially since I am able to send and receive messages.

Thanks for the help!

Juancams, 2025-05-22:
Hello!

Regarding your problem, I'll need a little more information. I don't know if you could specify what type of connection you have with the robot (Ethernet or WLAN), what DDS you have on your computer, what version of the repos you're using, and how you're posting (there's an issue, I don't know if it's open or closed, that lists the commands to move it).

> `[WARN] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/utlidar/mapping_cmd' with type 'std_msgs::msg::dds_::String_' from USER_DATA '(null)'. (handle_builtintopic_endpoint() at [...]

This warning confuses me, since I have seen it when trying to communicate versions of ros2 higher than jazzy with humble or lower, but not humble with foxy.

I look forward to your response.

Best regards ^^

16stelter, 2025-05-22:
Heya,

thanks for the quick response. I figured out the problem in the mean time: I accidentally locked the robot joints, so it was incapable of moving. Disabled the lock and now everything seems to work fine. The warning is weird, but it seems to be just a warning, so I'll simply ignore it for now as it doesn't impact any functionality.

The only thing I'm still a bit confused about is the nav2 part. I've tried sending a 2D goal pose through rviz, but that doesn't seem to do anything. I assume there are other nodes that need to be started for that to work? Right now I am only launching go2_bringup.

To answer your questions, I am connected to the robot through ethernet and using cyclonedds and the humble branch of this repo.